secile/UsbCamera #5: Can This Work With WPF

AlBannaTechno, 2020-10-09:
First Thankyou for this awesome work,

is there a way to use it with WPF?

secile, 2020-10-21:
Hello, thank you for your question.
I am not familiar with WPF, but as I tried,
this source code could be used if you add reference to 'System.Drawing'.

GetBitmap() returns System.Drawing.Bitmap, so you have to convert to BitmapSource
if you want to use in BitmapSource of Image control.

AlBannaTechno, 2020-10-22:
@secile thank you, I used another library but still use the same trick since that library also supports only windows form.

So your answer is 100% Correct, thank you.

closed 2020-10-26 by secile

yangjieshao, 2020-11-02:
@AlBannaTechno  
you can use like this 
https://github.com/secile/UsbCamera/pull/6