secile/UsbCamera #7: Camera preset

artmanxxx, 2020-12-07:
Hi,
How i call/set preset from/to cam?
Example
tnx

secile, 2020-12-10:
Hello. Are you saying that you want to get default value of  tilt, zoom, exposure, brightness, contrast, and so on?

artmanxxx, 2020-12-10:
Hi,
First, thank you for your answer.
No, 
I know how to get default value of tilt, zoom.

I don't know how i get current value, and how i set or call preset in/from camera memory. Over UVC usb
If i connect camera to RS232 i send command **VISCA**

81 01 04 00 02 FF - On
81 01 04 00 03 FF - Off

81 01 04 3D 02 FF - CAM WDR On
81 01 04 3D 03 FF - CAM WDR Off

81 01 04 33 02 FF - BlackLight On
81 01 04 33 03 FF - BlackLight Off

Call preset 1/2
81 01 06 01 VV WW **01** 03 FF
81 01 06 01 VV WW **02** 03 FF

I am work in IT department high school. We have 100 hybrid class ( camera and mic )
I need create small program for control ptz camera.

[image]

secile, 2020-12-11:
Are you meaning the word 'preset' is a set of tilt, zoom, exposure, brightness..., that is saved on camera memory, and you want to switch these properties at once? My library use DirectShow, and I do not know such method in DirectShow.

To get current value, you have to change source code a little. could you try this?

line 254: current
```C#
public Property(int min, int max, int step, int @default, int flags, Action<DirectShow.CameraControlFlags, int> set)
{
    this.Min = min;
    this.Max = max;
    this.Step = step;
    this.Default = @default;
    this.Flags = (DirectShow.CameraControlFlags)flags;
    this.CanAuto = (Flags & DirectShow.CameraControlFlags.Auto) == DirectShow.CameraControlFlags.Auto;
    this.SetValue = set;
    this.Available = true;
}
```

is change to

```C#
public Func<int> GetValue { get; private set; }
public Property(int min, int max, int step, int @default, int flags, Action<DirectShow.CameraControlFlags, int> set, Func<int> get)
{
    this.Min = min;
    this.Max = max;
    this.Step = step;
    this.Default = @default;
    this.Flags = (DirectShow.CameraControlFlags)flags;
    this.CanAuto = (Flags & DirectShow.CameraControlFlags.Auto) == DirectShow.CameraControlFlags.Auto;
    this.SetValue = set;
    this.GetValue = get;
    this.Available = true;
}
```

line 201: current

```C#
prop = new Property(min, max, step, def, flags, (flag, value) => cam_ctrl.Set(item, value, (int)flag));
```

is change to

```C#
Func<int> get = () =>
{
    int value = 0;
    cam_ctrl.Get(item, ref value, ref flags);
    return value;
};
prop = new Property(min, max, step, def, flags, (flag, value) => cam_ctrl.Set(item, value, (int)flag), get);
```

line 208: current
```C#
prop = new Property(min, max, step, def, flags, (flag, value) => vid_ctrl.Set(item, value, (int)flag));
```

is change to

```C#
Func<int> get = () =>
{
    int value = 0;
    vid_ctrl.Get(item, ref value, ref flags);
    return value;
};
prop = new Property(min, max, step, def, flags, (flag, value) => vid_ctrl.Set(item, value, (int)flag), get);
```

now, you can use
```C#
var value = prop.GetValue();
```

Please let me know If it works fine. If it is OK, I will commit my repository.

artmanxxx, 2020-12-14:
Hi,
Ok, thank you. Will check and let you know.

Best regards,
Aleksey Danilov

artmanxxx, 2020-12-14:
Thank, work!!!
I not really professional programmer. I am system IT

How i can optimize this code? 

`
        public void GetTiltValue()
        {
            UsbCamera.VideoFormat[] formats = UsbCamera.GetVideoFormat(comboListCameraBox.SelectedIndex);
            var CameraCreateConnect = new UsbCamera(comboListCameraBox.SelectedIndex, formats[0]);

            UsbCamera.PropertyItems.Property propTilt;
            propTilt = CameraCreateConnect.Properties[DirectShow.CameraControlProperty.Tilt];
            var value = propTilt.GetValue();

            richTextBox1.Text = Convert.ToString(value);
        }
`

secile, 2020-12-16:
Hello, thank you for your response.
As to your code, I could not comment because too less information. Typically, you had better create UsbCamera instance once in Form constructor or Load event.

artmanxxx, 2020-12-16:
secile very thank.

closed 2020-12-16 by artmanxxx