secile/UsbCamera #8: How to black out the camera system wide?

JannesStroehlein, 2021-02-24:
Hello,

I want to black out the camera. I had decent success with setting the brightness to the min value but this seems to be not working for every camera model. Can I apply a DirectShow Filter without "starting" the camera (I cannot start it if another application is already using it)? If so: What DirectShow filter should be used to make the camera stream go black?

Do you have any other suggestions?

Thanks in advance

Jannes

secile, 2021-02-25:
Hello, thank you for your question.  

Do you need to use DirectShow filter?
My suggestion is, how about change bitmap's brightness after GetBitmap() from UsbCamera.
See example below.

```csharp
private void Form_Load(object sender, EventArgs e)
{
    var camera = new GitHub.secile.Video.UsbCamera(0, new Size(640, 480));
    camera.Start();

    // trackBar1 value range is -100 to +100.
    float brightness_offset = 0;
    trackBar1.ValueChanged += (s, ev) => brightness_offset = trackBar1.Value / 100f;

    var timer = new System.Timers.Timer(100);
    timer.Elapsed += (s, ev) =>
    {
        var bmp = camera.GetBitmap();
        bmp = ChangeBitmapBrightness(bmp, brightness_offset);
        pbxScreen.Image = bmp;
    };
    timer.Start();

    this.FormClosing += (s, ev) =>
    {
        camera.Stop();
    };
}

// offset value range is -1 to +1.
// -1 becomes src image black out.
// +1 becomes src image white out.
// 0 do nothing.
private Bitmap ChangeBitmapBrightness(Bitmap src, float offset)
{
    var matrix = new System.Drawing.Imaging.ColorMatrix()
    {
        Matrix00 = 1, Matrix11 = 1, Matrix22 = 1, Matrix33 = 1, Matrix44 = 1,
        Matrix40 = offset, // r
        Matrix41 = offset, // g
        Matrix42 = offset, // b
    };

    var attribute = new System.Drawing.Imaging.ImageAttributes();
    attribute.SetColorMatrix(matrix);

    var result = new Bitmap(src.Width, src.Height);
    using (var g = Graphics.FromImage(result))
    {
        g.DrawImage(src, new Rectangle(Point.Empty, src.Size), 0, 0, src.Width, src.Height, GraphicsUnit.Pixel, attribute);
    }

    return result;
}
```

JannesStroehlein, 2021-02-25:
But does this work if another application is already using the camera? I need to be able to "black out" the camera for every application on the computer.

secile, 2021-02-25:
Oh I see. It can not work.  

I would like to confirm. Does another application means application that uses this UsbCamera library? or the application like camera app on Windows10?

JannesStroehlein, 2021-02-25:
I mean any other application for example Zoom, Skype, Discord etc. I had decent success with setting the camera's brightness to zero but it does only work on some cameras.

secile, 2021-02-25:
Thank you for your information.  
In my understandings, if you insert DirectShow filter, do not make effect to another application. Because each application make it's own DirectShow filter graph.  The change of filter graph make effect to your application only.

JannesStroehlein, 2021-02-25:
Ok thank you very much for you information and time. That would mean the best way would be a virtual camera device, which usually just passes the original camera through but can interrupt this stream to black the camera out. Is it possible to write this in C# or do I have to do is in C++?

secile, 2021-02-26:
I think using virtual camera is good idea. To make virtual camera, you have to create DirectShow video capture source filter in C++. Making DirectShow filter has some complex steps... so please google.  

I have ever been used following source code called 'ycapture' to make Virtual Camera. (no english, sorry)  
http://yzwlab.net/ycapture/

If you are going to use ycapture, I can share you how to send image to ycapture from your application.

closed 2021-03-31 by secile