secile/UsbCamera #11: Expand camera image

Elgamily, 2022-01-31:
How to make the camera image size be the same size of the PictureBox?

secile, 2022-02-01:
Hello, Thank you for your question.

You can specify camera image size by UsbCamera constructor 2nd parameter.
But supported image size is depend on your camera hardware.
You can know supported size by UsbCamera.GetVideoFormat( ).

To change image size same with PictureBox, you have to change bitmap size like this.

```cs
var org_bmp = camera.GetBitmap();
var new_bmp = new Bitmap(org_bmp, pictureBox1.Size);
```

Elgamily, 2022-03-04:
> I don't understand what problem happens. Please explain more detail and
> show me your code and image that has problem.

Hello,

Look at both photos the 1st photo is the photo that is created by the
liberary and 2nd photo is the original photo. There is a difference between
both of them. The photo that is created by the liberary doesn’t draw all of
the pixels in the original photo. What should I do?

Best regards,
Hassan Elgamily

secile, 2022-03-04:
Thank you for your reply. But, I think there is no attached photo file. Could you confirm it?

Elgamily, 2022-03-04:
No, it's not. I’ve attached two screenshots.

Elgamily, 2022-03-04:
[image]
[image]
Sorry, these are the two photos.

secile, 2022-03-04:
1st photo is the photo that is created by camera.GetBitmap()?  
And 2nd photo is captured by another app (for example, Windows Camera app)?

secile, 2022-03-04:
I think 2nd photo's aspect ratio is different from 1st photo. Could you check VideoFormat your camera supports?
Execute below and let me know result.

```cs
// check format.
int cameraIndex = 0;
UsbCamera.VideoFormat[] formats = UsbCamera.GetVideoFormat(cameraIndex);
for(int i=0; i<formats.Length; i++) Console.WriteLine("{0}:{1}", i, formats[i]);
```

Elgamily, 2022-03-07:
These are the results:

0:[Video], [YUY2], {Width=640, Height=480}, 333333, [VideoInfo],
Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={640,
480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=24576000, MaxBitsPerSecond=147456000,

1:[Video], [YUY2], {Width=640, Height=480}, 333333, [VideoInfo2],
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={640,
480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=24576000, MaxBitsPerSecond=147456000,

2:[Video], [YUY2], {Width=160, Height=120}, 333333, [VideoInfo],
Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={160,
120}, MinCroppingSize={160, 120}, MaxCroppingSize={160, 120},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={160, 120}, MaxOutputSize={160, 120}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=1536000, MaxBitsPerSecond=9216000,

3:[Video], [YUY2], {Width=160, Height=120}, 333333, [VideoInfo2],
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={160,
120}, MinCroppingSize={160, 120}, MaxCroppingSize={160, 120},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={160, 120}, MaxOutputSize={160, 120}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=1536000, MaxBitsPerSecond=9216000,

4:[Video], [YUY2], {Width=176, Height=144}, 333333, [VideoInfo],
Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={176,
144}, MinCroppingSize={176, 144}, MaxCroppingSize={176, 144},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={176, 144}, MaxOutputSize={176, 144}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=2027520, MaxBitsPerSecond=12165120,

5:[Video], [YUY2], {Width=176, Height=144}, 333333, [VideoInfo2],
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={176,
144}, MinCroppingSize={176, 144}, MaxCroppingSize={176, 144},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={176, 144}, MaxOutputSize={176, 144}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=2027520, MaxBitsPerSecond=12165120,

6:[Video], [YUY2], {Width=320, Height=176}, 333333, [VideoInfo],
Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={320,
176}, MinCroppingSize={320, 176}, MaxCroppingSize={320, 176},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={320, 176}, MaxOutputSize={320, 176}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=4505600, MaxBitsPerSecond=27033600,

7:[Video], [YUY2], {Width=320, Height=176}, 333333, [VideoInfo2],
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={320,
176}, MinCroppingSize={320, 176}, MaxCroppingSize={320, 176},
CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={320, 176}, MaxOutputSize={320, 176}, OutputGranularityX=1,
[... 590 lines cut ...]
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0,
InputSize={1280, 720}, MinCroppingSize={1280, 720}, MaxCroppingSize={1280,
720}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={1280, 720}, MaxOutputSize={1280, 720}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=73728000, MaxBitsPerSecond=442368000,

74:[Video], [MJPG], {Width=1280, Height=960}, 333333, [VideoInfo],
Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0,
InputSize={1280, 960}, MinCroppingSize={1280, 960}, MaxCroppingSize={1280,
960}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={1280, 960}, MaxOutputSize={1280, 960}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=98304000, MaxBitsPerSecond=589824000,

75:[Video], [MJPG], {Width=1280, Height=960}, 333333, [VideoInfo2],
Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0,
InputSize={1280, 960}, MinCroppingSize={1280, 960}, MaxCroppingSize={1280,
960}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1,
MinOutputSize={1280, 960}, MaxOutputSize={1280, 960}, OutputGranularityX=1,
OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0,
ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000,
MinBitsPerSecond=98304000, MaxBitsPerSecond=589824000,

secile, 2022-03-07:
How did you create UsbCamera class?
According to your photo1, photo1's asspect ratio might be 4:3, so you might choose 640x480?
If so, you might create UsbCamera class like this?

```c#
var camera = new UsbCamera(cameraIndex, formats[0]);
```

photo2' aspect ratio may be 16:9, so you have to choose VideoFormat of 16:9.
For example formats[32] is for example 1280x720. Please test below.

```c#
var camera = new UsbCamera(cameraIndex, formats[34]);
```

Elgamily, 2022-03-08:
> Did you read my two previous messages? Are you try below?
>
> var camera = new UsbCamera(cameraIndex, formats[34]);
>
> I think your problem's reason is difference of aspect rate. If this is not
> the solution of your problem, I have nothing any other idea.

It’s working after trying this. Many thanks for your help.

secile, 2022-03-09:
It is very nice. thank you.

closed 2022-03-09 by secile