secile/UsbCamera #13: Camera preview locked

Onafets51, 2022-03-20:
Using a The Imaging Source USB 3.1 1440 x 1080 px camera sometimes, especially at the beginning, the preview freezes and you have to do Camera.Stop() + Camera.Start() to make it restart. What could it be?

secile, 2022-03-22:
Hello, Thank you for your question.

1. As I described in comment in line 32, immediately after starting the USB camera,
GetBitmap() fails because image buffer is not prepared yet.
What you are saying is this situation?

2. If not,
I don't know if it works, but could you try the following change around line 318?
And please let me know it work fine or not.

before change.
```cs
public int BufferCB(double SampleTime, IntPtr pBuffer, int BufferLen)
{
    if (Buffer == null || Buffer.Length != BufferLen)
    {
        Buffer = new byte[BufferLen];
    }

    lock (BufferLock)
    {
        Marshal.Copy(pBuffer, Buffer, 0, BufferLen);
    }
    return 0;
}
```

after change.
```cs
public int BufferCB(double SampleTime, IntPtr pBuffer, int BufferLen)
{
    lock (BufferLock)
    {
        if (Buffer == null || Buffer.Length != BufferLen)
        {
            Buffer = new byte[BufferLen];
        }

        Marshal.Copy(pBuffer, Buffer, 0, BufferLen);
    }
    return 0;
}
```

Onafets51, 2022-03-22:
Thanks for your suggestion, I modified the code as shown. The defect seems less frequent, moreover it does it especially when the camera is cold and it happens also with AMCAP. With their test program IC Capture (The Imagin Source) it never happens.

secile, 2022-03-24:
Hello.
I found that it is necessary to return quickly in callback function.
if callback does not return quickly, it can interfere with playback.
https://docs.microsoft.com/en-us/windows/win32/directshow/isamplegrabber-setcallback

This is previous version document, it is prohibited to enter critical section.
https://docs.microsoft.com/en-us/previous-versions/ms786692(v=vs.85)

Could you try this version?
I changed do not enter critical section and do not wait another thread.
And please let me know your problem is solved.

```cs
public int BufferCB(double SampleTime, IntPtr pBuffer, int BufferLen)
{
    if (Buffer == null || Buffer.Length != BufferLen)
    {
        Buffer = new byte[BufferLen];
    }

    var locked = false;
    try
    {
        System.Threading.Monitor.TryEnter(BufferLock, 0, ref locked);
        if (locked)
        {
            Marshal.Copy(pBuffer, Buffer, 0, BufferLen);
        }
    }
    finally
    {
        if (locked) System.Threading.Monitor.Exit(BufferLock);
    }
    return 0;
}
```

Onafets51, 2022-03-24:
I also tried an old version of UsbCamera where there was no callback function but there was GetBitmapMain and GetBitmapMainMain, but even with that same problem. Waiting for the new camera-board to arrive to draw conclusions.

secile, 2022-03-24:
Thank you for your response.

One last thing I want to check.
Please execute following code and check output message.

```cs
// check format.
int cameraIndex = 0;
UsbCamera.VideoFormat[] formats = UsbCamera.GetVideoFormat(cameraIndex);
for(int i=0; i<formats.Length; i++) Console.WriteLine("{0}:{1}", i, formats[i]);
```

for example, this is my result.
```
0:[Video], [MJPG], {Width=1280, Height=720}, 333333, [VideoInfo], ...
1:[Video], [MJPG], {Width=320, Height=180}, 333333, [VideoInfo], ...
2:[Video], [MJPG], {Width=320, Height=240}, 333333, [VideoInfo], ...
...
9:[Video], [YUY2], {Width=1280, Height=720}, 1000000, [VideoInfo], ...
10:[Video], [YUY2], {Width=320, Height=180}, 333333, [VideoInfo], ...
11:[Video], [YUY2], {Width=320, Height=240}, 333333, [VideoInfo], ...
...
18:[Video], [RGB24], {Width=1280, Height=720}, 1000000, [VideoInfo], ...
19:[Video], [RGB24], {Width=320, Height=180}, 333333, [VideoInfo], ...
20:[Video], [RGB24], {Width=320, Height=240}, 333333, [VideoInfo], ...
```

Now, if you are using formats[0], (that is [MJPEG]),

```cs
var camera = new UsbCamera(0, formats[0]);
```

please use same size but another format like formats[9], (that is [YUY2]) or formats[18], (that is [RGB24])?

Onafets51, 2022-03-24:
[image]

I hope you can see

Onafets51, 2022-03-24:
I use format #5

secile, 2022-03-25:
I have one question. When you used format #5 (Y800), aside from occasional preview freeze, are you able to get correct image? (unbroken image?)

BufferToBitmap function works correctly if image memory layout is RGB24. But you are using Y800 format and it's memory layout is difference from RGB24.

Onafets51, 2022-03-25:
Hi,
Yes with Y800 format #5 and also with Y16 format #1 I had a correct image. However, after the latest changes (format RGB24 format #9), on my development machine (CPU Intel Core i7 8 core 3.8 GHz) the problem did not occur anymore but, on a tablet with CPU Celeron Gemini Lake N4120 4 core 2.4 GHz instead it manifested itself. I'm waiting to receive the new camera to do more tests.

secile, 2022-03-25:
Thank you. I'm feeling little strangeness, but I accept the real.
I guess samplegrabber (parts of DirectShow) coverts Y16/Y800 to RGB24 appropriately.

I close this issue. If you have any question, ask me again, but I have not so much idea any more.

closed 2022-03-25 by secile