secile/UsbCamera #20: How can I get the datas of every frame?

hellottyp, 2022-11-28:


secile, 2022-11-28:
Hello.
You have to modify UsbCamera.cs a little.
Try these changes below and let me know it works fine or not.

Step1. Add this block around line 80.

```cs
public Action<Bitmap> BitmapCaptured
{
    get { return CaptureGrabberCallback.Buffered; }
    set { CaptureGrabberCallback.Buffered = value; }
}
private SampleGrabberCallback CaptureGrabberCallback;
```

Step2. Replace this block from

```cs
private Func<Bitmap> GetBitmapFromSampleGrabberCallback(DirectShow.ISampleGrabber i_grabber, int width, int height, int stride)
{
    var sampler = new SampleGrabberCallback(width, height, stride);
    i_grabber.SetCallback(sampler, 1); // WhichMethodToCallback = BufferCB
    return () => sampler.GetBitmap();
}
```

to

```cs
private Func<Bitmap> GetBitmapFromSampleGrabberCallback(DirectShow.ISampleGrabber i_grabber, int width, int height, int stride)
{
    CaptureGrabberCallback = new SampleGrabberCallback(width, height, stride);
    i_grabber.SetCallback(CaptureGrabberCallback, 1); // WhichMethodToCallback = BufferCB
    return () => CaptureGrabberCallback.GetBitmap();
}
```

Step3. subscribe BitmapCaptured.

```cs
camera.BitmapCaptured += (bmp) =>
{
    // called here every frame captured.
};
```

hellottyp, 2022-11-29:
good,thanks!it works very well.

hellottyp, 2022-11-29:
it works,but it isn't thread-safe.

secile, 2022-11-29:
Could you show me the code that produce the problem?

hellottyp, 2022-11-29:
like this:
```
 camera.BitmapCaptured += (bmp) =>
  {
      this.textBox1.Text = "hello";
      this.BackgroundImage = bmp;
  };
```
You're better off implementing it with events

secile, 2022-11-29:
Are you saying that the line below cause InvalidOperationException? (Cross-thread operation not valid.)
```cs
this.textBox1.Text = "hello";
```

If so, it is your responsibility to call cross-thread control in a thread-safe way.

```cs
textBox1.Invoke((Action)(() =>
{
    this.textBox1.Text = "hello";
}));
```

hellottyp, 2022-11-29:
I did it.And it can work.But I have many controls to process.So I will do more work.It's not just convenient.

hellottyp, 2022-11-29:
I use delegates to handle those controls, so I do a lot of things. Using Invoke is much easier.thanks!

closed 2022-11-29 by secile