secile/UsbCamera #21: missing events

hellottyp, 2022-12-07:
``
            // create instance.
            camera = new UsbCamera(cameraIndex, format);
            // this closing event handler make sure that the instance is not subject to garbage collection.
            this.FormClosing += (s, ev) => camera.Release(); // release when close.

            // show preview on control. (works light.)
            camera.SetPreviewControl(this.panVideo.Handle, this.panVideo.ClientSize);

        private void panVideo_MouseMove(object sender, MouseEventArgs e)
        {
            //it never fired over the camera region.but fired over other whitespace region.
            Console.WriteLine("mouse move");
        }

secile, 2022-12-08:
Hello, thank you for your information.

This is library's problem.
I'll fix future release, but could you add 1 line (see below) and try again?

```cs
// about line 260.
builder.RenderStream(ref pinCategory, ref mediaType, vcap_source, null, null);
vw.put_Owner(controlHandle);
vw.put_MessageDrain(controlHandle);   <- ADD THIS LINE.
```

hellottyp, 2022-12-08:
panel1.MouseMove += (s, ev) =>
                    {
                        Console.WriteLine("mouse move"); // work well
                    };

                    panel1.MouseLeave += (s, ev) =>
                    {
                        Console.WriteLine("mouse leave"); //fired too
                    };

                    panel1.MouseClick += (s, ev) =>
                    {
                        Console.WriteLine("mouse click"); //not work
                    };

hellottyp, 2022-12-08:
I have a none border window.I want to drag the window by the container of the camera.

secile, 2022-12-09:
It seems that it is limitation of DirectShow.
Please use MouseUp and MouseDown event instead.

closed 2023-01-11 by secile