secile/UsbCamera #23: provide nuget package

dojo90, 2023-02-01:
First of all I would like to thank you for the library! Really impressive and kept simple 👍

Finally, to use your code, I added the [UsbCamera.cs](https://github.com/secile/UsbCamera/blob/master/UsbCamera.cs) to my project. It would be nice if you could also provide us with a nuget package (taking versioning into account; see https://github.com/GitTools/GitVersion).

wxharry, 2024-05-27:
Hi secile, I would like to help if you haven't worked on the nuget packging. 😄

secile, 2024-06-09:
Thank you.
This issue is currently stopped because I felt some difficulties when creating DLL.
The library change behavior if it is defined 'USBCAMERA_WPF' symbol that makes GetBitmap() returns image of BitmapSource.
To release the library as nuget package, I have to create DLL, but it is not able to do same behavior with DLL.

Maybe I have to create two kind of DLL like UsbCameraWinForms and UsbCameraWpf.
and I have to provide 2 kind of nuget package. (Is my idea correct?)

Could you give me more time to think it over.

wxharry, 2024-06-09:
> Thank you. This issue is currently stopped because I felt some difficulties when creating DLL. The library change behavior if it is defined 'USBCAMERA_WPF' symbol that makes GetBitmap() returns [...]

Hi, thank you for the reply. I think we can have adapters for wpf and winforms that implement the GitBitmap() differently and resue the other codes. To use the dll(let's say it is named as Secile.UsbCamera), developers can use base methods by `using Secile.UsbCamera`, and import wpf related methods by `using Secile.UsbCamera.Wpf` and the same for the winforms `using Secile.UsbCamera.Winforms`. What do you think?