secile/UsbCamera #25: 需要增加对于部分非标的虚拟摄像头的过滤(It is necessary to increase the filtering of some non-standard virtual cameras)

yangjieshao, 2023-03-12:
1. 下载并安装 非标的寻摄像头(例如 YY开播）
(Download and install non-standard cameras  such as 'YY开播')
[YY开播](https://113-204-230-30-adb9862a.ksyungslb.com:8043/yydl.yy.com/4/setup/YYSetup-9.8.0.0-zh-CN.exe)

2.  FindDevices 成功获取到 ‘YY开播’
(find 'YY开播’ success by FindDevices  )
[image]

3. GetVideoFormat 时报错
(error on GetVideoFormat )
[image]

造成这个的原因是 ‘YY开播’ 注册了一个信息不完整的USB摄像头
(The reason for this is that a USB camera with incomplete information has been registered for 'YY开播')

问题是我们没办法限制用户不要安装什么软件
(The problem is that we can't restrict users from installing any software)

我发现在 GetFiltes 时   'YY开播' 无法获取到 DevicePath
(I found that 'YY开播' could not get DevicePath when GetFiltes)
[image]

所以 尝试获取 DevicePath  或许是一个过滤非标虚拟摄像头的方式
(So trying to get DevicePath may be a way to filter non-standard virtual cameras)

----------------
我发现 [FlashCap ](https://github.com/kekyo/FlashCap) 并没有这个问题，他直接就过滤了非标的虚拟USB摄像头信息
(I found that [FlashCap ](https://github.com/kekyo/FlashCap) did not have this problem. It directly filtered the non-standard virtual USB camera information)

yangjieshao, 2023-03-14:
仅仅获取 FriendlyName 是不报异常的
(Only getting FriendlyName is not an exception)
只有在获取 DevicePath 的时候会报异常
(An exception will be reported only when obtaining DevicePath)

事实上 这确实是'YY开播'在实现USB摄像头时并未安装标准实现产生的bug
(In fact, this is really a bug caused by 'YY开播' not installing standard implementation when implementing USB camera)

但是我们并不能保证所有的第三方厂商的实现都是标准的
(However, we cannot guarantee that all third-party implementations are standard)

对于那些不标准的实现，我们至少要做到不暴露异常给最终用户
(For those non-standard implementations, we should at least not expose exceptions to end users)

secile, 2023-08-25:
Hello, sorry for late reply.

To tell the truth, I couldn’t understand what is a problem.
I tried to download ‘YY开播’, but I can’t use it because I can’t read Chinese.
But, I finally understand.

You are saying that, virtual camera such as ‘YY开播’ successes FindDevices,
But do not successes GetVideoFormat and do not work correctly. So it should not be shown by FindDevices.

As to this problem, I will not filter ‘YY开播’ to disable, but let ‘YY开播’ works in default video format.
* when calling GetVideoFormat was failed, GetVideoFormat returns special value ‘VideoFormat.Default’.
* when you pass VideoFormat.Default to UsbCamera constructor, UsbCamera do not change video format.

Could you test it?
[UsbCameraIssue#25.zip](https://github.com/secile/UsbCamera/files/12434195/UsbCameraIssue.25.zip)

closed 2023-08-25 by yangjieshao

wxharry, 2024-01-04:
I have a related problem with the virtual camera, I installed [obs](https://obsproject.com/) on my machine and it is detected as a device but it cannot work properly.

wxharry, 2024-01-04:
> I have a related problem with the virtual camera, I installed [obs](https://obsproject.com/) on my machine and it is detected as a device but it cannot work properly.

I found get DevicePath throws an error on the virtual camera which is a way to filter it out. However, it throws an error. Alternatively, you can add [PreserveSig] to the read method and compare the output to filter out virtual cameras.

secile, 2024-01-10:
@wxharry 
Hello, thank you for your information.
I confirmed that obs virtual camera do not work with my library, but Microsoft Teams app is able to use obs virtual camera.
I do not know why, but In my estimation, obs virtual camera must be used with obs studio and app that uses virtual camera at a same time,
but my library is based on DirectShow, and DirectShow can not use camera at a same time from different app.
Teams may not use DirectShow, but use more modern technology, I think.