secile/UsbCamera #26: Problem with showing preview

RickyMartine, 2023-04-26:
First of all, thank you for this. This is realy great.

I'm trying to develop wpf app with this and i was testing with 3 different cameras (Logitech brio, one from lg and razer kiyo).
Lg and Razer are working fine with no problem, but i cant get it to work with brio and i dont know why.
I tried it with all possible video formats that i get from GetVideoFormat() method.

Problem appears here, when it tries to subscribe to event:
[image]

 
[image]

Message of exception here is: "Value does not fall within the expected range."
[image]

Any idea, what could be or can be done? Thanks.

secile, 2023-04-26:
Hello. Thank you for using my library and sending me a question.

I don't know reason why, but it seems that your camera Logitech brio is not support 'Preview Pin'.
As far as I know, even if camera do not support 'Preview Pin', 
DirectShow's component Capture Graph Builder will make 'Preview Pin' automatically
by inserting Smart Tee filter. (This is described in comment line 292 and below.)
So, I believed all camera supports 'Preview Pin'.

As a plan B, is it possible to adopt '2. use Timer and GetBitmap()' approach?

secile, 2023-04-26:
If you can use GraphEdit, I have some items I would like you to try.
GraphEdit is tool for test DirectShow created by Microsoft.
As to GraphEdit, show page below. 

https://learn.microsoft.com/en-us/windows/win32/directshow/using-graphedit?ranMID=46128&ranEAID=wizKxmN8no4&ranSiteID=wizKxmN8no4-1yUTJ264eGGZk13yddGnyQ&epi=wizKxmN8no4-1yUTJ264eGGZk13yddGnyQ&irgwc=1&OCID=AID2200057_aff_7794_1243925&tduid=(ir__ib09dtywykkfbyq9klrj3irq2n2x6r6bpdcwhd6d00)(7794)(1243925)(wizKxmN8no4-1yUTJ264eGGZk13yddGnyQ)()&irclickid=_ib09dtywykkfbyq9klrj3irq2n2x6r6bpdcwhd6d00

secile, 2023-04-26:
By using timer, you got black image, right?
Something is wrong...

Does another app can use Logitech BRIO? for instance, windows camera app can use the camera?

in terms of image of GraphEdit, there are 2 Capture Pin. maybe, the camera has 2 dual cameras like in camera and out camera? If so, That may be the reason.

LucienMP, 2025-11-08:
I also got the same problem with NZXT 4K60 device, and same code path.

There is a chance this code returns but doesnt fill the Dictionary.
https://github.com/secile/UsbCamera/blob/master/UsbCamera.cs#L128-L133

What about adding an else to the following code?
https://github.com/secile/UsbCamera/blob/master/UsbCamera.cs#L392-L406

```
  Releasing += () => Streams[StreamType.Preview].Release();
}else {
  throw new NotSupportedException("Camera probably doesnt support preview pin.");
}

```

secile, 2025-11-12:
I have got logicool MX BRIO 705 and test it.
This is a dump of VideoFormat.

VideoFormat that subType is [YUY2] or [MJPG] works fine.
But VideoFormat that subType is {3231564e-0000-0010-8000-00aa00389b71} (that represents 'NV12'} do not work,
and happens same problem!

```
0:[Video], [YUY2], {Width=640, Height=480}, 166666, 60fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={640, 480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=294912000, MaxBitsPerSecond=294912000, 
1:[Video], [YUY2], {Width=640, Height=480}, 166666, 60fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={640, 480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=294912000, MaxBitsPerSecond=294912000, 
2:[Video], [YUY2], {Width=160, Height=120}, 166666, 60fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={160, 120}, MinCroppingSize={160, 120}, MaxCroppingSize={160, 120}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={160, 120}, MaxOutputSize={160, 120}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=18432000, MaxBitsPerSecond=18432000, 
3:[Video], [YUY2], {Width=160, Height=120}, 166666, 60fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={160, 120}, MinCroppingSize={160, 120}, MaxCroppingSize={160, 120}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={160, 120}, MaxOutputSize={160, 120}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=18432000, MaxBitsPerSecond=18432000, 
4:[Video], [YUY2], {Width=176, Height=144}, 166666, 60fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={176, 144}, MinCroppingSize={176, 144}, MaxCroppingSize={176, 144}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={176, 144}, MaxOutputSize={176, 144}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=24330240, MaxBitsPerSecond=24330240, 
5:[Video], [YUY2], {Width=176, Height=144}, 166666, 60fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={176, 144}, MinCroppingSize={176, 144}, MaxCroppingSize={176, 144}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={176, 144}, MaxOutputSize={176, 144}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=24330240, MaxBitsPerSecond=24330240, 
6:[Video], [YUY2], {Width=320, Height=180}, 166666, 60fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={320, 180}, MinCroppingSize={320, 180}, MaxCroppingSize={320, 180}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={320, 180}, MaxOutputSize={320, 180}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=55296000, MaxBitsPerSecond=55296000, 
7:[Video], [YUY2], {Width=320, Height=180}, 166666, 60fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={320, 180}, MinCroppingSize={320, 180}, MaxCroppingSize={320, 180}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={320, 180}, MaxOutputSize={320, 180}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=55296000, MaxBitsPerSecond=55296000, 
8:[Video], [YUY2], {Width=320, Height=240}, 166666, 60fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={320, 240}, MinCroppingSize={320, 240}, MaxCroppingSize={320, 240}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={320, 240}, MaxOutputSize={320, 240}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=166666, MaxFrameInterval=2000000, MinBitsPerSecond=73728000, MaxBitsPerSecond=73728000, 
9:[Video], [YUY2], {Width=320, Height=240}, 166666, 60fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={320, 240}, MinCroppingSize={320, 240}, MaxCroppingSize={320, 240}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={320, 240}, MaxOutputSize={320, 240}, OutputGra [... cut ...]
89:[Video], 3231564e-0000-0010-8000-00aa00389b71, {Width=2560, Height=1440}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={2560, 1440}, MinCroppingSize={2560, 1440}, MaxCroppingSize={2560, 1440}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={2560, 1440}, MaxOutputSize={2560, 1440}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=2000000, MinBitsPerSecond=1327104000, MaxBitsPerSecond=1327104000, 
```

secile, 2025-11-12:
@LucienMP
could you tell me what VideoFormat are you using?

If you are goint to use 'NV12', please use another VideoFormat such as [YUY2], [MJPG], or [RGB24]
for the time being.

It will take some time to decide whether to support 'NV12'.

LucienMP, 2025-11-22:
> [@LucienMP](https://github.com/LucienMP) could you tell me what VideoFormat are you using?
> 
> If you are goint to use 'NV12', please use another VideoFormat such as [YUY2], [MJPG], or [RGB24] for the time being.
> 
> It will take some time to decide whether to support 'NV12'.

Yes, but also i am finding that for the NZXT that the line :
```
var sample = ConnectSampleGrabberAndRenderer(graph, builder, vcap_source, DirectShow.DsGuid.PIN_CATEGORY_CAPTURE, format);
if ( sample != null ) ... // <------- always null for the NZXT
```

This is a problem because it quietly fails, so all other dictionary entry are zero