secile/UsbCamera #28: How to distinguish the camera with the same name?

Wissdom09, 2023-08-01:
Your code is very good.
But I have confirmed some problems.

Currently implemented code is to get a camera list, select a camera (select an index value), and connect.
But What I want is to save the name/ID of the camera and extract the matching index value.
and than I want to connect after extracting the index value. (Automatic connection)
(The index of each camera is changed every time because the camera can be changed, broken, or newly added.
It cannot be used as a specification because the camera cannot be selected every time)

At first, 
I tried to use the camera name and [VideoFormat] as specifications 
(Find the camera name and the  [VideoFormat] Find the index value)
However, when using the cameras of the same product, Since the name is the same, the method cannot be used.

I want an ID like Aforge.video.directShow -Filtermoniker, 
In the Videoo format [] GetVideoFormat (int Cameraidex) function, m_symboliclik is identified as Aforge.video.directShow -Filtermoniker.
but it could not be used becuase it is IUNKNOWN type

Is it possible to add m_symboliclink value to the [VideoFormat] for distinction between cameras?
Can you tell me if there is a better way?

Thank you very much

yangjieshao, 2023-08-01:
you can read `DevicePath`

```
/// <summary>
/// 获取过滤器列表
/// </summary>
/// <param name="category"></param>
/// <param name="isVideo">是否视频设备</param>
/// <param name="isWave">是否音频设备</param>
/// <returns></returns>
public static List<DeviceInfo> GetFiltes(Guid category, bool isVideo, bool isWave)
{
	var result = new List<DeviceInfo>();

	EnumMonikers(category, (moniker, prop) =>
	{
		try
		{
			DeviceInfo info = new DeviceInfo();
			object value = null;
			prop.Read("FriendlyName", ref value, 0);
			info.Name = value + string.Empty;
			if (isVideo)
			{
				prop.Read("DevicePath", ref value, 0);
				info.DevicePath = (string)value;
			}
			if (isWave)
			{
				prop.Read("WaveInID", ref value, 0);
				info.WaveInID = value + string.Empty;
			}

			result.Add(info);
		}
		catch
		{
			// can not find Virtual Cam 
		}

		return false; // 継続。
	});

	return result;
}
```

Wissdom09, 2023-08-01:
你的写作救了我。
非常感谢

secile, 2023-08-01:
Thank you for your question.

I agree with @yangjieshao 's comment.
Thank you @yangjieshao.

Please read issue #3.

yangjieshao, 2023-08-05:
> 你的写作救了我。 非常感谢

所以  你为啥 还不关掉这个issues

closed 2023-08-06 by secile