secile/UsbCamera #29: Add a feature to get plain image data as byte array to not call lock bits over and over again

Mrgaton, 2023-08-06:
I'm making a movement detector with your awesome class but going through image GetPixel and SetPixel is very slow so it would be great if that could be possible

Mrgaton, 2023-08-06:
And maybe add like an event when a camera new frame is captured? and wait for the event to end before sending the next one

secile, 2023-08-08:
Thank you for your request.

I can't decide right now whether I will comply with your request.
For the time being, using GetPixel/SetPixel is very slow. So you had better use LockBits instead of that.
See example below. Is this can be solution of your problem?

```cs

private byte[] ExtractBitmapData(Bitmap bmp)
{
    var rect = new Rectangle(Point.Empty, bmp.Size);
    var data = bmp.LockBits(rect, System.Drawing.Imaging.ImageLockMode.ReadOnly, System.Drawing.Imaging.PixelFormat.Format32bppArgb);
    var bytes = Math.Abs(data.Stride) * bmp.Height;
    var result = new byte[bytes];
    System.Runtime.InteropServices.Marshal.Copy(data.Scan0, result, 0, result.Length);
    return result;
}
```

secile, 2023-08-08:
As to an event when a new frame is captured.

Are you using WinForms, or WPF?
If WinForms, You can subscribe PreviewCaptured callback.
(If WPF, PreviewCaptured is only use for show preview with DataBinding.)

 Is this can be solution of your problem?

```cs
camera.PreviewCaptured += (bmp) =>
{
    // called when every preview captured.
};
```

yangjieshao, 2023-08-09:
[image]
Perhaps you want to directly obtain this `Buffer` to reduce unnecessary bitmap conversions?

secile, 2023-08-09:
Hello, @yangjieshao.
Maybe, you are right.
But, I'm not sure if this feature is necessary for many other users.
And I think it's difficult to switch between Func<Bitmap> GetBitmap and Func<byte[]> GetBitmap to implement this functionality.

@Mrgaton 
For now, I make your custum UseCamera.cs. Could you try this?
GetBitmap() returns byte[] (not Bitmap.)
[UsbCameraForIssue#29.zip](https://github.com/secile/UsbCamera/files/12299071/UsbCameraForIssue.29.zip)

Mrgaton, 2023-08-09:
Or maybe add a function to get buffer instead of get bitmap

secile, 2023-08-13:
@Mrgaton

I recently sent your custom version of UsbCamera.cs last day.
Is it satisfied your requirement?

Mrgaton, 2023-08-13:
Can you add like a link like a credit on the start of the class with the link to you github repository

Mrgaton, 2023-08-13:
> @Mrgaton
> 
> I recently sent your custom version of UsbCamera.cs last day. Is it satisfied your requirement?

yes but as i said i think it would be good  to have two functions one to get bitmap and another to get it as byte array

Mrgaton, 2023-08-13:
like 

Camera.GetBitmapRaw()

and 

Camera.GetBitmap()

Mrgaton, 2023-08-13:
and how do i know the fps of the library that captures?

yangjieshao, 2023-08-14:
> and how do i know the fps of the library that captures?

you can get `VideoFormat` by `UsbCamera.GetVideoFormat(cameraIndex)`
You can calculate it yourself based on `VideoFormat.TimePerFrame`
[image]

secile, 2023-08-22:
@Mrgaton
> Can you add like a link like a credit on the start of the class with the link to you github repository

Thank you, I will do.

> Camera.GetBitmapRaw()
> Camera.GetBitmap()
> cause i need both

This library has several ways to get image data.
* GetBitmap()
* PreviewCaptured
* StillImageCaptured

I wouldn't add a function to get a buffer just to GetBitmap() only.
To do this, I will add new conditional compilation symbol to switch type of image data.
Converting byte array to Bitmap is easy.

```cs
private static Bitmap BufferToBitmap(byte[] buffer, int width, int height)
{
    var result = new Bitmap(width, height);
    var bmpData = result.LockBits(new Rectangle(Point.Empty, result.Size), System.Drawing.Imaging.ImageLockMode.WriteOnly, System.Drawing.Imaging.PixelFormat.Format24bppRgb);
    System.Runtime.InteropServices.Marshal.Copy(buffer, 0, bmpData.Scan0, buffer.Length);
    result.UnlockBits(bmpData);
    return result;
}
```

secile, 2023-08-22:
I modified UsbCameraForIssue#29. Could you test it?
[UsbCameraForIssue#29_Rev2.zip](https://github.com/secile/UsbCamera/files/12403539/UsbCameraForIssue.29_Rev2.zip)

* fix problem that byte array image data is upside down.
* add new conditional compilation symbol 'USBCAMERA_BYTEARRAY' to change image type from Bitmap to byte array.

When you define conditional compilation symbol 'USBCAMERA_BYTEARRAY',
GetBitmap returns byte array instead of Bitmap. (and PreviewCaptured, StillImageCaptured passes data of byte array.)

To be precise, GetBitmap returns IEnumerable\<byte\>.
The data type is actually a byte array, so cast it to byte array before use it.

```cs
button1.Click += (s, ev) =>
{
    var buf = (byte[])camera.GetBitmap();
    var bmp = BufferToBitmap(buf, camera.Size.Width, camera.Size.Height);
    pictureBox2.Image = bmp;
};

private static Bitmap BufferToBitmap(byte[] buffer, int width, int height)
{
    var result = new Bitmap(width, height);
    var bmpData = result.LockBits(new Rectangle(Point.Empty, result.Size), System.Drawing.Imaging.ImageLockMode.WriteOnly, System.Drawing.Imaging.PixelFormat.Format24bppRgb);
    System.Runtime.InteropServices.Marshal.Copy(buffer, 0, bmpData.Scan0, buffer.Length);
    result.UnlockBits(bmpData);
    return result;
}
```

If you have any questions, please ask.

closed 2023-08-30 by secile