secile/UsbCamera #31: Inconsistent Image Quality When Saving Still Capture Monochrome Sensor Data as BMP

SankarGaneshKumar, 2023-12-11:
We are experiencing a discrepancy in the image quality when saving Still capture from a monochrome sensor as an 8-bit BMP file. The output image does not match the expected quality and format, differing significantly from a reference BMP image known to be of high quality.

secile, 2023-12-12:
Hello, thank you for your information.
I have never used monochrome sensor.

1. What product are you using? please send product url, if possible.
1. you said 'Still capture' is discrepancy.  Is it means image from GetBitmap() is good, and image from StillImageCaptured is no good?
1. Are you using WinForms, or WPF?
1. Send me the image if possible.
1. Let me know dump of VideoFormat. To get dump, execute as follows.

```cs
// get video format.
var cameraIndex = 0;
var formats = UsbCamera.GetVideoFormat(cameraIndex);

// select the format you want.
foreach (var item in formats) Console.WriteLine(item);
```

SankarGaneshKumar, 2023-12-13:
Thank you for your prompt response.

1. I have attached a zip file containing all the details you requested. It includes the datasheet for the cameras we are currently using.
2. Both the Still Capture and GetBitmap() functions yield the same quality, which is lower compared to the reference image taken from the camera manufacturer's app. We attempted to save the images in various formats such as .png, .bmp, and .tif. However, the size of the image increases without a corresponding improvement in quality.
3. Our application utilizes WinForms.

4 & 5. Please find the relevant information in the attachments.

Note: When attempting to obtain the video feed using your code, we initially encountered an error stating that it was unable to create a filter. To address this, I commented out the code as shown below (1st way):

// To show the preview, there are 3 ways.
// 1. Use SetPreviewControl. (Works smoothly, recommended.)
camera.SetPreviewControl(pictureBox1.Handle, pictureBox1.ClientSize);
pictureBox1.Resize += (s, ev) => camera.SetPreviewSize(pictureBox1.ClientSize); 

// 2. Use Timer and GetBitmap().
/*
var timer = new System.Timers.Timer(1000 / 30) { SynchronizingObject = this };
timer.Elapsed += (s, ev) => pictureBox1.Image = camera.GetBitmap();
timer.Start();
this.FormClosing += (s, ev) => timer.Stop();
*/

[Data.zip](https://github.com/secile/UsbCamera/files/13657815/Data.zip)

secile, 2023-12-14:
Could you send me 2 more information.

(6) When you create UsbCamera instance, which video format are you use?
In case the code below, I selected formats[0].

```cs
// get video format.
var cameraIndex = 0;
var formats = UsbCamera.GetVideoFormat(cameraIndex);

// select the format you want.
foreach (var item in formats) Console.WriteLine(item);
var format = formats[0]; // <- selected format[0].

// create instance.
var camera = new UsbCamera(cameraIndex, format);
```

(7) What is a value of bBitCount of bitmap header member about line 446.
[image]

SankarGaneshKumar, 2024-01-04:
Hi, 

For 6) we are using 55 CUG Camera , videoFormat[3] 
For 7th question share the above screenshot
[VideoFormats.txt](https://github.com/secile/UsbCamera/files/13826108/VideoFormats.txt)
[image]

secile, 2024-01-08:
At first I thought it was because biBitCount was 8.
The BmpBuilder.BufferToBitmap function does not work correctly if the buffer has 8 bits.
But this time it wasn't the cause.

Now you are using videoFormat[3].
What happen if you use another VideoFormat? especially, use VideoInfo instead of VideoInfo2.

SankarGaneshKumar, 2024-01-16:
Tried both VideoInfo and VideoInfo2, but both are giving same quality. Attached the saved images for reference and here is the code which saves the still capture image

`try
            {
                // Check if a frame was previously captured
                if (bmp != null)
                {
                    Image img = (Image)bmp;
                    // Save the frame to a file
 
 
                    Image image2 = Image.FromHbitmap(bmp.GetHbitmap());
                    image2.Save(path, System.Drawing.Imaging.ImageFormat.Jpeg;
                    // Display a message or perform any desired action
                    //  MessageBox.Show("Frame captured and saved!");
                }
            }
            catch (Exception ex)
            {
                Console.WriteLine("Camera Snapsot exception : " + DeviceName + ":" + ex.Message);
            }`

Do you think the code which is saving still capture is causing the issue?
            
            
[image]

[image]

secile, 2024-01-18:
>Do you think the code which is saving still capture is causing the issue?

StillImageCapture works fine in my environment.

The GetBitmap function and StillImageCapture call the same function.
If GetBitmap is working fine, I think it is unlikely that StillImageCapture will work not fine.

One possibility, I want to confirm.
As to 'Reference_Image.bmp' and 'Using_Still_Pin Capture.jpg' files you sent me,
'Reference_Image.bmp' is bitmap file, that is not compressed, and 
'Using_Still_Pin Capture.jpg' is jpg file, that is compressed.
Could you save 'Using_Still_Pin Capture' to bitmap, and compare with 'Reference_Image.bmp'?

'Using_Still_Pin Capture.jpg' may have been compressed and degraded.

To save bitmap instance to file, please see code below.
I think you do not need to convert Bitmap to Image.

```cs
camera.StillImageCaptured += bmp =>
{
    bmp.Save(path, System.Drawing.Imaging.ImageFormat.Bmp);
};
```

rajeshamudala, 2024-06-20:
Thanks for the response ,  we used the above code but still we are having quality issues , I am attaching image where left is reference image and right is the image we got from above code , Still i feel the left image have more quality
[image]