secile/UsbCamera #32: Can't able to set brightness and exposure at run time

rajeshamudala, 2024-01-16:
Hi , 
I am running this to run 2 camera's in my project. When i tried to adjust exposure and brightness when i am creating camera it was working , when when i try to adjust brightness or contrast at run time using slider or like any other way it was throwing exception , I attached related screen shot. But when i ran the camera on main thread it was brightness and contrast are adjusting. But I need to run each camera on different thread. How to make sure that Brightness and contrast can adjust  when camera's are running on different threads.
[image]

secile, 2024-01-17:
Hello, thank you for your question.
As you said, it is true that it happens exception if you accessed camera instance from different thread create camera instance.

To handle this situation, I think it's depend on your program architecture,
but one solution is, use event Set() and WaitOne().

Here is sample.
```cs
private void Form1_Load(object sender, EventArgs e)
{
    System.Threading.ThreadPool.QueueUserWorkItem(x =>
    {
        // here is worker thread. create camera instance here.
        UsbCamera camera = CreateCamera();

        var value_changed_event = new System.Threading.AutoResetEvent(false);
        int value = 0;
        numericUpDown1.ValueChanged += (s, ev) =>
        {
            // here is UI thread. to access camera instance, use event.Set().
            value = (int)numericUpDown1.Value;
            value_changed_event.Set();
        };

        // main loop.
        while (true)
        {
            // wait until value changed.
            value_changed_event.WaitOne();
            // resume thread if event.Set() invoked.

            // set property.
            var prop = camera.Properties[DirectShow.CameraControlProperty.Focus];
            if (prop.Available)
            {
                prop.SetValue(DirectShow.CameraControlFlags.Manual, value);
            }
        }
    });
}
```

rajeshamudala, 2024-06-20:
Thank you it's worked for me!!!

closed 2024-06-20 by rajeshamudala