secile/UsbCamera #35: How can I turn off the usbcamera  lights?

fhdream, 2024-03-08:
When the usbcamera has lights,the lights will be turned on 
after executing code {***moniker.BindToObject(null, null, ref guid, out value)***}. 

But the light is not turned off after {*** usbcamera.Relase().*** }. 
How can I turn off the lights?

secile, 2024-03-08:
Hello.
On my PC, when I execute camera.Stop(), turned off the light.
camera.Release() execute camera.Stop() internal.

fhdream, 2024-03-08:
> Hello.
> 
> On my PC, when I execute camera.Stop(), turned off the light.
> 
> camera.Release() execute camera.Stop() internal.

fhdream, 2024-03-08:
> Hello.
> 
> On my PC, when I execute camera.Stop(), turned off the light.
> 
> camera.Release() execute camera.Stop() internal.

Thank you for your reply.

camera.Release() execute camera.Stop() internal.

But on my PC the light always on.
I don't know how to turn off the light.

yangjieshao, 2024-03-08:
It may be caused by unreasonable hardware design.
My solution is to dismantle the LED light

secile, 2024-03-09:
On my PC, your problem is not occur, so I can't test.
But, please test the code below. What could happen.

line 418.

```cs
Releasing += () => Stop();
```

to

```cs
Releasing += () =>
{
    vcap_source.Stop();
    Stop();
};
```

On my pc, light turned off at the line 'vcap_source.Stop();'.

fhdream, 2024-03-11:
> 1. Are you saying that light do not turned off after your app quit?
> 2. On my PC, your problem is not occur, so I can't test.
>    But, please test the code below. What could happen.
> 
> line 418.
> 
> ```cs
> Releasing += () => Stop();
> ```
> 
> to
> 
> ```cs
> Releasing += () =>
> {
>     vcap_source.Stop();
>     Stop();
> };
> ```
> 
> On my pc, light turned off at the line 'vcap_source.Stop();'.

1. The light must be turned off when exiting the app. But I need a button to control the light off.
2. I added the code but still can't turn off the light.
` Stop = () => DirectShow.PlayGraph(graph, DirectShow.FILTER_STATE.Stopped);
 Releasing += () =>
 {
     vcap_source.Stop();
     Stop();
 };
 Released += () =>
 {
     DirectShow.ReleaseInstance(ref builder);
     DirectShow.ReleaseInstance(ref graph);
 };
 Properties = new PropertyItems(vcap_source);`

secile, 2024-03-11:
If you want to turn off the light while using camera, it is impossible.

fhdream, 2024-03-11:
> If you want to turn off the light while using camera, it is impossible.

What I mean is using the button to turn off the usbcamera and turn off the lights at the same time.
Now the only way to turn off the camera and lights at the same time is to exit the entire application process.

secile, 2024-03-11:
OK, I understands.
On my PC, when I execute camera.Stop(), turned off the light.
I'm sorry, but I do not have any idea.

fhdream, 2024-03-11:
> OK, I understands. On my PC, when I execute camera.Stop(), turned off the light. I'm sorry, but I do not have any idea.

 Thanks for replying every time.
 I solved the problem, but I Do not execute function GetPinCategory. 
I don’t know if it will affect anything.

What I did was as follows
1.Do not execute this code `private static Guid GetPinCategory(IPin pPin)`
2.Released += () =>
{
          Add Codes
          DirectShow.ReleaseInstance(...);//The fuction GetVideoFormat()=>basefilter&&pin
          DirectShow.ReleaseInstance(...);//The fuction CreateVideoCaptureSource()=>basefilter&&pin
          DirectShow.ReleaseInstance(...);//The fuction SetVideoOutputFormat()=>config
}

closed 2024-12-26 by secile