secile/UsbCamera #48: USB Camera Latency Issue

buddlecity, 2025-05-12:
Hi,
I'm displaying a 3264x2448 image from a USB camera in real-time using a PictureBox, but there is more than a 2-second video delay.
When testing with OpenCV, the delay is under 1 second.
The current implementation reads frames in a thread and outputs them to the PictureBox.
I'd appreciate any support or suggestions on how to resolve this issue.

secile, 2025-05-12:
Hello. thank you for your question.

could you execute the line below and
tell me the result of Console.WriteLine(item).

```cs
foreach (var item in formats) Console.WriteLine(item);
```
 
There is a Fps field in the dump.
I want to check this value.

buddlecity, 2025-05-12:
[Video], [MJPG], {Width=800, Height=600}, 333333, 30fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={800, 600}, MinCroppingSize={800, 600}, MaxCroppingSize={800, 600}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={800, 600}, MaxOutputSize={800, 600}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=230400000, MaxBitsPerSecond=230400000, 

[Video], [MJPG], {Width=800, Height=600}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={800, 600}, MinCroppingSize={800, 600}, MaxCroppingSize={800, 600}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={800, 600}, MaxOutputSize={800, 600}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=230400000, MaxBitsPerSecond=230400000, 

[Video], [MJPG], {Width=640, Height=480}, 333333, 30fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={640, 480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=147456000, MaxBitsPerSecond=147456000, 

[Video], [MJPG], {Width=640, Height=480}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={640, 480}, MinCroppingSize={640, 480}, MaxCroppingSize={640, 480}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={640, 480}, MaxOutputSize={640, 480}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=147456000, MaxBitsPerSecond=147456000, 

[Video], [MJPG], {Width=1280, Height=720}, 333333, 30fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={1280, 720}, MinCroppingSize={1280, 720}, MaxCroppingSize={1280, 720}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1280, 720}, MaxOutputSize={1280, 720}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=442368000, MaxBitsPerSecond=442368000, 

[Video], [MJPG], {Width=1280, Height=720}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={1280, 720}, MinCroppingSize={1280, 720}, MaxCroppingSize={1280, 720}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1280, 720}, MaxOutputSize={1280, 720}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=442368000, MaxBitsPerSecond=442368000, 

[Video], [MJPG], {Width=1600, Height=1200}, 333333, 30fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={1600, 1200}, MinCroppingSize={1600, 1200}, MaxCroppingSize={1600, 1200}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1600, 1200}, MaxOutputSize={1600, 1200}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=921600000, MaxBitsPerSecond=921600000, 

[Video], [MJPG], {Width=1600, Height=1200}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={1600, 1200}, MinCroppingSize={1600, 1200}, MaxCroppingSize={1600, 1200}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1600, 1200}, MaxOutputSize={1600, 1200}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=921600000, MaxBitsPerSecond=921600000, 

[Video], [MJPG], {Width=1920, Height=1080}, 333333, 30fps, [VideoInfo], Guid=05589f80-c356-11ce-bf01-00aa0055595a, VideoStandard=0, InputSize={1920, 1080}, MinCroppingSize={1920, 1080}, MaxCroppingSize={1920, 1080}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1920, 1080}, MaxOutputSize={1920, 1080}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=995328000, MaxBitsPerSecond=995328000, 

[Video], [MJPG], {Width=1920, Height=1080}, 333333, 30fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={1920, 1080}, MinCroppingSize={1920, 1080}, MaxCroppingSize={1920, 1080}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1920, 1080}, MaxOutputSize={1920, 1080}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=333333, MaxFrameInterval=333333, MinBitsPerSecond=995328000, MaxBitsPerSecond=995328000, 

[... cut ...]
[Video], [YUY2], {Width=1920, Height=1080}, 2000000, 5fps, [VideoInfo2], Guid=f72a76a0-eb0a-11d0-ace4-0000c0cc16ba, VideoStandard=0, InputSize={1920, 1080}, MinCroppingSize={1920, 1080}, MaxCroppingSize={1920, 1080}, CropGranularityX=1, CropGranularityY=1, CropAlignX=1, CropAlignY=1, MinOutputSize={1920, 1080}, MaxOutputSize={1920, 1080}, OutputGranularityX=1, OutputGranularityY=1, StretchTapsX=0, StretchTapsY=0, ShrinkTapsX=0, ShrinkTapsY=0, MinFrameInterval=2000000, MaxFrameInterval=2000000, MinBitsPerSecond=165888000, MaxBitsPerSecond=165888000,

secile, 2025-05-12:
You may use VideoFormat this one?

```
[Video], [MJPG], {Width=3264, Height=2448}, 666666, 15fps
```

In my experience, fps value may reliable...

secile, 2025-05-12:
Could you show me your source code?

buddlecity, 2025-05-12:
I'm using [Video], [MJPG], {Width=3264, Height=2448}, 666666, 15fps.

The video is displayed in the PictureBox from a thread or timer at roughly 30ms intervals.
The video does not freeze and is displayed continuously, but there is about a 2-second delay in the output.

buddlecity, 2025-05-12:
using System;
using System.Collections.Generic;
using System.ComponentModel;
using System.Data;
using System.Drawing;
using System.Linq;
using System.Text;
using System.Windows.Forms;

using System.IO;
using System.Runtime.Serialization;
using GitHub.secile.Video;

namespace USBCamera_Test2
{
    public partial class Form1 : Form
    {
        public string[] CamDevices = null;
        private UsbCamera mCamera = null;

        public Form1()
        {
            InitializeComponent();
        }

        private void Form1_Load(object sender, EventArgs e)
        {
            CamDevices = UsbCamera.FindDevices();
            UsbCamera.VideoFormat[] formats = UsbCamera.GetVideoFormat(0);
            mCamera = new UsbCamera(0, formats[14]);
            mCamera.Start();

            pictureBox1.SizeMode = PictureBoxSizeMode.Zoom;
            timer1.Interval = 30;
            timer1.Enabled = true;
            
        }

        private void Form1_FormClosed(object sender, FormClosedEventArgs e)
        {
            timer1.Enabled = false;
            mCamera.Stop();
            mCamera.Release();
        }

        private void timer1_Tick(object sender, EventArgs e)
        {
            if (mCamera.IsReady)
            {
                if (pictureBox1.Image != null)
                {
                    pictureBox1.Image.Dispose();
                    GC.Collect();
                }
                pictureBox1.Image = mCamera.GetBitmap();
            }
        }
    }
}

secile, 2025-05-13:
I checked your code, and there is no suspicious code. my library include sample project. is same problem happens if you use sample project?

secile, 2025-05-13:
(2) just in case, please tell me your PC spec.
CPU, RAM size, OS?

(3) what happens if you chance VideoFormat more smaller size? does improve frame rate?

buddlecity, 2025-05-14:
1) Yes, the same issue occurs. There's about a 2-second delay in the output.

2) i7-8550 / 8GB / Windows 10

3) There's almost no delay at 1280x720 resolution.

4) I'll check it.

5) It's a Chinese-made camera, so there's no specific model name. However, the delay is under 1 second when using OpenCvSharp.

As a reference, after modifying the code as below, the performance is almost similar to OpenCvSharp, though OpenCvSharp is still slightly faster.

1. Fixed the Buffer[] size to about 4K. Added a BufferLength variable and used it for the buffer copy operation.

2. Changed SubType to MEDIASUBTYPE_MJPG.

3. In the GetBitmap() function, converted MJPG to BMP for usage.

I still want to reduce the delay further.
Thank you..

secile, 2025-05-18:
Only for preview, it is recommended not use 'timer and GetBitmap()', but you had better to use 'SetPreviewControl()' in WinForms.

It is more less memory usage.
please see SampleProject to know how to use SetPreviewControl().

buddlecity, 2025-05-19:
Thank you for your help.

closed 2025-05-19 by buddlecity