secile/UsbCamera #49: The USB camera cannot update the picture after hot-plugging.

andyhebear, 2025-07-10:
The USB camera cannot update the picture after hot-plugging.Usb相机热拔插后不能更新图片

yangjieshao, 2025-07-10:
https://github.com/Bassman2/UsbMonitor

你要这玩意？

andyhebear, 2025-07-11:
like this:`      
while(true) {
                                    System.Threading.Thread.Sleep(30);
                                    using(var bmp = m_usbCamera.GetBitmap()) {
                                        //do some thing
                                        ...
                                    }
                                 }
`
i find when this usb  camera device  Unplug it and then plug it into the computer. the var bmp = m_usbCamera.GetBitmap() cann't update

andyhebear, 2025-07-11:
and when i m_usbCamera.Release() . after i find thead no  stop in  the ref SampleGrabberCallback ;you need modify code in SampleGrabberCallback class Implement the interface IDisposable
`
  private class SampleGrabberCallback : DirectShow.ISampleGrabberCB, IDisposable
        {
            private byte[] Buffer;
            private object BufferLock = new object();

            public ActionHandler<Bitmap> Buffered { get; set; }

            private System.Threading.AutoResetEvent BufferedEvent;

            private BitmapBuilder BmpBuilder;

            public bool IsReady { get { return Buffer != null; } }
            internal bool IsDisposed { get { return _isDisposed; }private set { _isDisposed = true; } }
            private volatile bool _isDisposed = false; // volatile保证多线程可见性        
            private System.Threading.Thread m_thread;
            public SampleGrabberCallback(DirectShow.ISampleGrabber grabber, int width, int height, int stride, bool useCache) {
                this.BmpBuilder = new BitmapBuilder(width, height, stride, useCache);

                // create Buffered.Invoke thread.
                BufferedEvent = new System.Threading.AutoResetEvent(false);      
                // use new Thread instead of ThreadPool. (issue #30)
                m_thread = new System.Threading.Thread(x => {
                    while(!IsDisposed) {
                        //todo:
                        BufferedEvent.WaitOne(); // wait event.
                        Buffered?.Invoke(GetBitmap()); // fire!
                    }
                    //清理资源
                    BufferedEvent.Close();
                    BufferedEvent = null;
                });
                m_thread.IsBackground = true;
                m_thread.Start();

                grabber.SetCallback(this, 1); // WhichMethodToCallback = BufferCB
            }

            public Bitmap GetBitmap() {
                if(Buffer == null) return BmpBuilder.EmptyBitmap;

                lock(BufferLock) {
                    return BmpBuilder.BufferToBitmap(Buffer);
                }
            }

            // called when each sample completed.
            // The data processing thread blocks until the callback method returns. If the callback does not return quickly, it can interfere with playback.
            public int BufferCB(double SampleTime, IntPtr pBuffer, int BufferLen) {
                if(Buffer == null || Buffer.Length != BufferLen) {
                    Buffer = new byte[BufferLen];
                }

                // replace lock statement to Monitor.TryEnter. (issue #14)
                var locked = false;
                try {
                    locked = System.Threading.Monitor.TryEnter(BufferLock, 0);
                    if(locked) {
                        Marshal.Copy(pBuffer, Buffer, 0, BufferLen);
                    }
                }
                finally {
                    if(locked) System.Threading.Monitor.Exit(BufferLock);
                }

                // notify buffered to worker thread. (issue #16)
                if(Buffered != null) BufferedEvent.Set();

                return 0;
            }

            // never called.
            public int SampleCB(double SampleTime, DirectShow.IMediaSample pSample) {
                throw new NotImplementedException();
            }
            //增加销毁时退出线程
            public void Dispose() {
                _isDisposed = true;
                BufferedEvent?.Set(); // 触发终止事件
                m_thread?.Join(100);    // 等待线程退出
                if(m_thread.IsAlive) {
                    // 强制终止（最后手段）
                    m_thread.Abort();
                }
            }
        }
`

and in UsbCamera class you need modify method:
`
   /// <summary>Release resource.</summary>
        public void Release() {
            Releasing();
            Released();
            //强制结束内部开启的线程
            try {
                foreach(var v in Streams.Values) {
                    if(v != null&&v.IsDisposed==false) {
                        v.Dispose();
                    }
                }
                
            }
            catch {
            }
        }

`

secile, 2025-07-25:
I fixed the problem that 
do not finish worker thread in SampleGrabberCallback when camera.Release().

closed 2025-07-25 by secile